raspberrypi/debugprobe #150: for the love of god, can you be more verbose with your release info?

billj324, 2024-09-01:
for the love of god, can you be more verbose with your release info?  I'm a beginner and having 4 different firmware files without details next to them add so much time to these easter egg hunts.  It saves so many people time just to write a QUICK description of the files to save collective HUNDREDS of hours of having to differentiate them my self.  This is a problem I have community-wide.  Makes the learning curve so much more painful if all devs do is assume everyone knows what the hell is going on.

lurch, 2024-09-01:
See Appendix A. of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf and https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#updating-the-firmware-on-the-debug-probe

billj324, 2024-09-06:
first off, I want to appologize for venting my frustrations.  you dont deserve that.
I've followed everything in those PDFs verbatim and still... a (would you say...) plethora of issues.
Lets start out by listing my env:
Windows 11 w/msys64/mingw64.  I'm doing this completely outside of VScode, because there are a littany of issues with that, including it not passing my pico_w correctly (didnt wanna mess w it rn because I would rather avoid a layer and just use command lines.
I've struggled with environment variable like crazy with this one.
doing the typical export command to set the sdk path IS not only inconsistent, but flawed in mingw64, due to inconsistent behavior within same mingw session, vs reloading it, v.s. rebooting computer/reloading mingw
I've dug through mountains of documentation to find the most incomplete documentation on the issue or the behavior does NOT match documentation.  For instance, I still cant find where to permanently change the pico-sdk-path so that opening mingw64 has it set right, despite export -p showing it is in fact set right,
running cmake persistently tries to pick it up from pico-sdk-1.5.x.
setting it with export is only per-session.
I set it in the windows env, upon reboot mingw64 now lists it properly within export -p.
HOWEVER, cmake still doesnt get it.  SO:
 
cmake -DPICO_SDK_PATH="/c/pico/pico_sdk2" -DDEBUG_ON_PICO=ON -DPICO_BOARD=pico_w ..
this finally works, except....

C:/pico/debugprobe/build/generated/pico_base/pico/config_autogen.h:7:10: fatal error: /c/pico/pico_sdk2/src/boards/include/boards/pico_w.h: No such file or directory
    7 | #include "/c/pico/pico_sdk2/src/boards/include/boards/pico_w.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

I can, with great certainty, say the file DOES exist.
I get this even when specifying:
-DPICO_BOARD=pico, but complaining pico.h cannot be found.  I'm so close to making this work I can SMELL it.
this is becoming a head scratcher.  
Can I specify the header file directly, with say -DPICO_BOARD=/c/pico/pico_sdk2/src/board/include/pico_w.h?
or is there another issue I'm not understanding correctly?

kilograham, 2024-09-06:
Cc @will-v-pi

billj324, 2024-09-06:
as another update, vscode keeps trying to use ninja, but I followed the instructions for manually entering it into perferences>settings>cmake extension > cmake:generator  > "NMake Makefiles" w/o quotes

lurch, 2024-09-06:
> C:/pico/debugprobe/build/generated/pico_base/pico/config_autogen.h:7:10: fatal error: /c/pico/pico_sdk2/src/boards/include/boards/pico_w.h: No such file or directory 7 | #include [...]

I'm just guessing here (I don't use Windows myself), but I wonder if part of the problem might be that some part of your system is trying to use MinGW-style paths (`/c/pico/blah`) but another part of your system is trying to use Windows-style paths (`C:\pico\blah`) ? :man_shrugging:

billj324, 2024-09-06:
before going to sleep, I processed it for a bit, the conclusion I came to was the same, its having issues with the path interpolation.  I'm on windows by choice right now, this laptop is the most mobile machine I have so out of convenience, I'm using it.  I have two Linux-native machines, looks like I'm going to switch, as I see it being more sustainable with fewer issues. I was just hoping I could master this env, but the issues seem endless.  my final attempt at this will come after spending some time to master the difference to fill out my knowledge.  I will give the released firmware one more try, I was hoping to get the LED working so I had some sign of life from the picow,  if that fails, I'll swith to another (I have the pins soldered to eliminate that vector of failure). Thank you all.  not going to give up on this, despite the frustration, I still find the challenge fun

billj324, 2024-09-06:
got the debug output from openOCD.  Its not getting response from the target picow anyone spot anything?
[oocdoutpt.txt](https://github.com/user-attachments/files/16911801/oocdoutpt.txt)

billj324, 2024-09-06:
I have good news and bad news,  I was able to get it to receive output from the target picow, through the UART on the debugprobe picow.  I can see the outputs in both putty and the serial monitor now.  the issue with openocd appears to still remain.  This is no longer your issue I believe.  The problem now lies with openOCD.  Thank you all.

lurch, 2024-09-07:
> UPDATE: need someone to build for pico_w.

Are you happy to close this issue as a duplicate of #41 ?

Neril, 2024-10-08:
Please update the uf2 compilation instructions to reference Ninja as a possible build tool. I'm probably unusual in using Windows / VS Code, but it took me a while to work out that cmake had generated a set of ninja files, not the Makefile that the instructions assume, and therefore the instructions were not working.

As it happens, I was also building for Pico2, which didn't even mention running make after cmake, so I had incorrectly concluded that the "make" line above was a typo.

closed 2025-11-24 by P33M