raspberrypi/debugprobe #151: add new interface and target into openocd

rahmati6410, 2024-09-16:
Hi 
I am new at openocd. I want to program samL11E15A (programming via swd) in my custom board. I could program it with atmel studio using ATMEL-ICE hardware in windows. Now i want to program the chip by openocd and i do not know which /interface/? config file I have to use, or I have to write my openocd.cfg for config interface and target.

lurch, 2024-09-18:
You may have noticed the "raspberrypi" part of this repo's URL? :wink:  We only provide support for using DebugProbe with RP2040- and RP2350-based boards, not boards based on other microcontrollers.

closed 2024-09-18 by lurch

rahmati6410, 2024-09-22:
> You may have noticed the "raspberrypi" part of this repo's URL? 😉 We only

Hi
Thank you for your answer.