raspberrypi/debugprobe #152: Changing SWCLK and SWDIO gpios

ThePurpleOne, 2024-09-19:
I wanted to use a PICO as a debugprobe with GPIOs different from the original ones.

I changed them in the `board_pico_config.h`  file:

```c
#define PROBE_IO_RAW
// #define PROBE_CDC_UART

// PIO config
#define PROBE_SM 0
#define PROBE_PIN_OFFSET 2

// ! BASE CONFIG
// #define PROBE_PIN_SWCLK (PROBE_PIN_OFFSET + 0) // 2
// #define PROBE_PIN_SWDIO (PROBE_PIN_OFFSET + 1) // 3

// ! TEST CONFIG OK
#define PROBE_PIN_SWCLK (PROBE_PIN_OFFSET + 1) // GPIO 3
#define PROBE_PIN_SWDIO (PROBE_PIN_OFFSET + 2) // GPIO 4

// ! TEST NOT OK
// #define PROBE_PIN_SWCLK (PROBE_PIN_OFFSET + 2) // GPIO 4
// #define PROBE_PIN_SWDIO (PROBE_PIN_OFFSET + 3) // GPIO 5

// ! TEST NOT OK
// #define PROBE_PIN_SWCLK (PROBE_PIN_OFFSET + 12) // GPIO 14
// #define PROBE_PIN_SWDIO (PROBE_PIN_OFFSET + 13) // GPIO 15

// ! TEST NOT OK
// #define PROBE_PIN_SWCLK (PROBE_PIN_OFFSET + 14) // GPIO 16
// #define PROBE_PIN_SWDIO (PROBE_PIN_OFFSET + 15) // GPIO 17
```

I've tried different pairs of pins, always consecutive but only one pair worked, the others do NOT communicate at all on the SWD port.

Am i missing something ? or should these different GPIOs work.

Note:
It's correctly compiled with `cmake -DDEBUG_ON_PICO=ON ..` and 2 different configs work so it's correctly compiled.

lurch, 2024-09-19:
You might need to change the `PROBE_PIN_OFFSET`, rather than changing the numbers that you add to it? (because of the way that https://github.com/raspberrypi/debugprobe/blob/develop/src/probe.pio#L105 works)

closed 2024-09-19 by ThePurpleOne

lurch, 2024-09-19:
@P33M Perhaps it's worth adding a compile-time error if `(PROBE_PIN_SWDIO - PROBE_PIN_OFFSET)` is too big?