raspberrypi/debugprobe #153: Conclicting documentation - debugprobe on Pico 2

arc12, 2024-10-08:
A This page - https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#debugging-using-another-pico-series-device - states:
"Pico 2 is not yet supported by debuprobe_on_pico debugging software. "

B The getting started guide states :
"One Pico or Pico 2 can reprogram and debug another using the debugprobe firmware, which transforms the Pico or Pico 2
into a USB → SWD and UART bridge."

C. The README for this repo describes how to build debugprobe_on_pico for Pico 2.

Is "A" in need of revision?

lurch, 2024-10-08:
pinging @nathan-contino our resident documentarian.

nathan-contino, 2024-10-09:
Thanks for reporting this. You seem to be correct -- we need to update https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#debugging-using-another-pico-series-device to indicate that Pico 2 is now supported.