raspberrypi/debugprobe #155: SDK 2.1.0 and Debug probe

eshea3, 2024-11-27:
After installing the new SDK 2.1.0 , the Pico does not reset  and hangs after loading new elf file with openocd.  Tried downloading the source for debugprobe and recompiling and that results in an "failed to connect to multidrop rp2040".

lurch, 2024-11-27:
Have you tried the recently-released firmware update? https://github.com/raspberrypi/debugprobe/releases/tag/debugprobe-v2.1.0

eshea3, 2024-11-27:
Downloaded one this morning, but that looks newer.  Will try that now.
Thanks

eshea3, 2024-11-27:
Tried that and still doesn't reset
command is
 sudo openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter
speed 5000" -c "program soil.elf verify reset exit"

If I run the command

sudo openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter
speed 5000" -c "init; reset; exit"

the Pico resets.

I have tried this on multiple Picos and multiple projects.  Only a dingle
debug probe though

eshea3, 2024-11-27:
And I did delete the build directory and rebuild each project after
updating the debugprobe firmware

lurch, 2024-11-27:
Pinging @will-v-pi as IIRC he looked at openocd recently?

eshea3, 2024-11-27:
Update

The Pico is actually rebooting.  The projects I tested on had in common the
following startup code

 stdio_init_all();
 sleep_ms(2000);

Removing the second line resolved the issue.  I have no idea why that
common throws things into a tizzy.
Thanks for your help.

will-v-pi, 2024-11-27:
That’s a known issue with the 0.12.0 release of OpenOCD not working with sleeps (https://github.com/raspberrypi/pico-feedback/issues/428) - it’s since been fixed, so this is probably unrelated to the 2.1.0 release and you’re likely using the wrong OpenOCD

eshea3, 2024-11-27:
> sleeps (raspberrypi/pico-feedback#428

Thanks will look into it.  But been using the same setup w/o problem until
I installed 2.1.0.

Gerhard-Vienna, 2025-01-12:
Look here: https://github.com/raspberrypi/pico-sdk/issues/2132#
This is definitely a problem with the current 2.1.0 release. 
As a temporary solution, switch the sdk to the development branch.

eshea3, 2025-01-12:
> Look here: raspberrypi/pico-sdk#2132

Thanks I saw that earlier this week and already adjusted my call to openOCD

eshea3, 2025-01-12:
Adding  -c "set USE_CORE 0" to the openocd invocation worked for me - no
need to downgrade the SDK