raspberrypi/debugprobe #159: can't build pico2 due to freetos dependency

thatux, 2024-12-27:
```
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m33_gcc'
CMake Error at FreeRTOS_Kernel_import.cmake:88 (message):
  Directory '/home/user/debugprobe/freertos' does not contain
  a 'rp2350-arm-s' port here:
  portable/ThirdParty/Community-Supported-Ports/GCC/RP2350_ARM_NTZ
Call Stack (most recent call first):
  CMakeLists.txt:6 (include)
```

I think it's a regression introduced with https://github.com/raspberrypi/debugprobe/commit/0c84dd0dc105228129fd3593d043fdb1f9fb84ba

only the 2040 is upstream https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/ThirdParty/GCC/RP2040

the 2350 is still downstream https://github.com/raspberrypi/FreeRTOS-Kernel/tree/main/portable/ThirdParty/GCC/RP2350_ARM_NTZ

https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/682f05 is checked out on a git clone

lurch, 2024-12-28:
I did a bit of digging, and the RP2350 port is indeed upstream at https://github.com/FreeRTOS/FreeRTOS-Kernel-Community-Supported-Ports/tree/main/GCC/RP2350_ARM_NTZ which gets included into https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/ThirdParty via a submodule, so perhaps your submodules aren't up to date?

thatux, 2024-12-28:
Instructions from the readme don't work for me:
```
git clone https://github.com/raspberrypi/debugprobe
cd debugprobe
git submodule sync
git submodule update --init
mkdir build-pico2
cd build-pico2
cmake -DDEBUG_ON_PICO=1 -DPICO_BOARD=pico2 -DPICO_PLATFORM=rp2350 ../
```

That's how I got the first error

lurch, 2024-12-28:
Ahhh, I've just tried this myself in a fresh directory, and it seems you're right. I guess the instructions need to be modified to do `git submodule update --init --recursive` instead.

closed 2025-01-09 by P33M