raspberrypi/debugprobe #160: can't build pico_w

thatux, 2024-12-27:
```
cmake -DDEBUG_ON_PICO=1 -DPICO_BOARD=pico_w ..
Using PICO_SDK_PATH from environment ('/home/user/.pico-sdk/sdk/2.1.0')
PICO_SDK_PATH is /home/user/.pico-sdk/sdk/2.1.0
Target board (PICO_BOARD) is 'pico_w'.
Using board configuration from /home/user/.pico-sdk/sdk/2.1.0/src/boards/include/boards/pico_w.h
Defaulting platform (PICO_PLATFORM) to 'rp2040' based on PICO_BOARD setting.
-- Defaulting build type to 'Release' since not specified.
Defaulting compiler (PICO_COMPILER) to 'pico_arm_cortex_m0plus_gcc' since not specified.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m0plus_gcc'
Defaulting PICO_GCC_TRIPLE to 'arm-none-eabi'
-- The C compiler identification is GNU 13.3.1
-- The CXX compiler identification is GNU 13.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/user/.pico-sdk/toolchain/13_3_Rel1/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/user/.pico-sdk/toolchain/13_3_Rel1/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/.pico-sdk/toolchain/13_3_Rel1/bin/arm-none-eabi-gcc
Build type is Release
-- Found Python3: /usr/bin/python3 (found version "3.12.3") found components: Interpreter 
TinyUSB available at /home/user/.pico-sdk/sdk/2.1.0/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/user/.pico-sdk/sdk/2.1.0/lib/btstack
cyw43-driver available at /home/user/.pico-sdk/sdk/2.1.0/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /home/user/.pico-sdk/sdk/2.1.0/lib/lwip
Pico W Wi-Fi build support available.
mbedtls available at /home/user/.pico-sdk/sdk/2.1.0/lib/mbedtls
CMake Error at CMakeLists.txt:66 (error):
  Unknown CMake command "error".
```

I think it's a regression introduced with 
https://github.com/raspberrypi/debugprobe/commit/1752f2a61b4c70501c5664ea945735e1e021f2d0 CMakeLists.txt doesn't implement the `pico_w` case. Furthermore the `Unknown CMake command "error" is really vague, not sure if error() is really a cmake function? https://github.com/raspberrypi/debugprobe/commit/1752f2a61b4c70501c5664ea945735e1e021f2d0#diff-1e7de1ae2d059d21e1dd75d5812d5a34b0222cef273b7c3a2af62eb747f9d20aR66

lurch, 2024-12-28:
> I think it's a regression introduced with...

It's not really a regression, since `debugbprobe` wasn't supported on Pico W before that change either :wink:  See #41

> Furthermore the `Unknown CMake command "error" is really vague, not sure if error() is really a cmake function?

Yeah, looks like it should probably have used `message()` instead? https://cmake.org/cmake/help/latest/command/message.html

thatux, 2024-12-28:
Oh I thought the Pico W was supported based on https://github.com/raspberrypi/debugprobe/issues/41#issuecomment-1678862423

lurch, 2024-12-28:
> Oh I thought the Pico W was supported based on [#41 (comment)](https://github.com/raspberrypi/debugprobe/issues/41#issuecomment-1678862423)

You might be able to get it working by making some manual edits, but AFAIK we don't officially support `debugprobe` on Pico W, because there's not yet any code in `debugprobe` to control the Pico W's onboard LED.
If you look at https://github.com/raspberrypi/debugprobe/releases you can see that the officially-supported boards are [DebugProbe](https://www.raspberrypi.com/products/debug-probe/), [Pico 1](https://www.raspberrypi.com/products/raspberry-pi-pico/) (non-W) and [Pico 2](https://www.raspberrypi.com/products/raspberry-pi-pico-2/) (non-W).

st2000, 2025-01-06:
>  ... AFAIK we don't officially support debugprobe on Pico W, because there's not yet any code in debugprobe to control the Pico W's onboard LED.

So, just to be clear, aside from no on-board PICO LED activity, a PICO W can run debugprobe to debug code running on another PICO?

lurch, 2025-01-07:
> So, just to be clear, aside from no on-board PICO LED activity, a PICO W can run debugprobe to debug code running on another PICO?

I don't know, I've never tried - I always just use the DebugProbe hardware.

closed 2025-01-09 by P33M