raspberrypi/debugprobe #161: Probe speed stuck at very low frequency

snaiperskaya96, 2025-01-22:
Hi,
Those few past days i've been trying a combination of rp2040 and rp2350 as both host and guest (debugger & target).
I've got myself a mix of 4+ different 2040s, in both the official dev board and some other custom formats, spanning different revision. 

No matter what software im using (openocd, pyOCD, probe-rs), i'm stuck with a speed that ranges between 10khz and 30khz, with 10 being the one that consistently works and 30 being somewhat unreliable where it could fail at some point during the flashing process.

Unfortunately i dont have any other debugger available to confirm that it is in fact an issue with debugprobe, but i dont know what else to look for so i'm opening this issue in the hope that its something known and workaround-able. 

I've also tried with older revisions, going back to 1.0 but no difference at all.

P33M, 2025-01-22:
How are you connecting the probe to the target (electrically)?

snaiperskaya96, 2025-01-22:
> How are you connecting the probe to the target (electrically)?

I felt dumb for a moment thinking i did not try to solder wires directly from the probe to the target, as i only connected them via jumpers beforehand. But, then i tried to solder them and unfortunately im running into the exact same issue.

snaiperskaya96, 2025-01-25:
Sorry for the delay. I'm following the documentation and double checking it with the pin definitions in [board_pico_config.h](https://github.com/raspberrypi/debugprobe/blob/master/include/board_pico_config.h).

| Probe |      Target      | 
|----------|:-------------:|
| GPIO 2 |  SWCLK |
| GPIO 3 | SWDIO | 
| VSYS | VSYS | 
| GND | GND | 

[image]

[image]

P33M, 2025-02-05:
5MHz definitely isn't going to work with that sort of setup. 30kHz is a bit low but not unsurprising.
You have a large loop between SWD/SWCLK and the GND wire because the connections are separated. What's really dumb is that the RP2040-zero has a single ground pad and that has to be used as the return for both Pico W power and SWD line current.

Connect the centre GND pin on the SWD/SCLK header on the Pico W to the GND on the RP2040-zero. Twist the SWD/SWCLK wires together with the GND wire. It should go a bit faster.