raspberrypi/debugprobe #163: Latest commit causes UART to break

PMLavigne, 2025-02-19:
The latest commit (83e28db) has done something to the USB connection that is causing huge problems with the UART side of things on my Mac. When I try to connect to the device with picocom, it sits there for a good 15 seconds thinking about it, finally gives me a "terminal OK" message, then proceeds to not print anything even though I know there's data there. Then it dies:

```
% picocom -b 115200 /dev/tty.usbmodem2102
picocom v3.1

port is        : /dev/tty.usbmodem2102
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

<15 seconds later>

Type [C-a] [C-h] to see available commands
Terminal ready

<15-30 seconds later>

FATAL: read from port failed: Device not configured
term_exitfunc: reset failed for dev UNKNOWN: Device not configured
```

This behavior occurred regardless of the baudrate I tried or other settings I picked.

I know it's the latest commit that did it, because I checked out the commit immediately before it (3d58384) and it works completely fine with no issues. Not sure what happened, but at least there's a workaround.

Additional Info:
 - Host computer: Macbook Pro with M1 chip running MacOS 15.1 (24B83)
 - Probe: RPi Pico using debug-on-pico mode firmware compiled as outlined above
 - Pico SDK: 2.1.0

Let me know if you'd like any other information, thanks.

P33M, 2025-02-19:
Oops. Might have been a mangled commit, I was testing a few things at the same time. For now, use the tagged release until I find some time to investigate.

PMLavigne, 2025-02-20:
... I don't know how I completely forgot that "tagged releases" are a thing that exists this morning, but I guess I did. D'oh. Oh well, I hope this was at least somewhat helpful anyway.

Cpasjuste, 2025-03-13:
Hi,

I had the same problem on linux, was going silly with this (tried different pico, different host, etc...). I did also forgot that "tagged releases" are a think, but github issues too :D

Thanks for all the great work here !

PMLavigne, 2025-03-13:
At least I feel slightly less dumb now that someone else did it too 😀

P33M, 2025-03-13:
That's sufficient prompting for a revert as people are obviously building manually.

lurch, 2025-03-13:
@P33M https://github.com/raspberrypi/pico-setup/blob/master/pico_setup.sh#L99 also does a plain `git clone` without checking out a specific branch or tag.