raspberrypi/debugprobe #166: CPU Clock set for RP2350

geek981108, 2025-03-17:
The CPU clock is currently defined in `DAP_config.h` as follows:

```c
#define CPU_CLOCK 125000000U ///< Specifies the CPU clock frequency in Hz.
```

This value represents the clock speed for the RP2040.

There is not a value for RP2350

lurch, 2025-03-17:
Presumably this means that if you use `debugprobe_on_pico2.uf2` then the RP2350 will be "underclocked" to 125MHz? Does this cause any problems?

geek981108, 2025-03-17:
> Presumably this means that if you use debugprobe_on_pico2.uf2 then the RP2350 will be "underclocked" to 125MHz?
That's right

> Does this cause any problems?
Not exactly, but it does result in a lower adapter speed. When debugging an RP2350 with a clock speed of 150 MHz, the adapter speed can be set to 30 MHz. However, when running at a clock speed of 125 MHz, the adapter speed only operates at around 20 MHz.

PR #174 (2025-05-14): probe: correct SWCLK calculations
Use the actual clk_sys frequency and round divisors up, otherwise high swclk speeds get significantly overclocked.

fixes #166 
closes #167
closed 2025-05-14 by P33M