raspberrypi/debugprobe #170: binaries?

petaflot, 2025-03-27:
project description states *Firmware source and binaries for the RP2040-based Debug Probe accessory* ; I still have to find the binary..

P33M, 2025-03-27:
Binaries are release artefacts: https://github.com/raspberrypi/debugprobe/releases

P33M, 2025-03-27:
The documentation for the Debug Probe is linked in the README and maintained at https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#debugging-using-another-raspberry-pi-pico
and this contains all of the steps required to flash the UF2 firmware, make the correct connections, and start OpenOCD debug sessions.

P33M, 2025-03-27:
Of course the docs site has churned since the Readme was updated... now links directly.

closed 2025-03-27 by P33M

petaflot, 2025-03-27:
oh right. my remark still stands, and I suggest adding

```
# Binaries

Binaries are release artefacts: https://github.com/raspberrypi/debugprobe/releases
```

above the TODO section in README.md