raspberrypi/debugprobe #173: Couldn't connect via swd to the stm microcontroller

Eksol, 2025-05-05:
The SWD bus is not working properly. I uploaded the debugprobe firmware to my RP2040, connected the STM32F401 via the SWD bus to the GPIO2 and GPIO3 pins on the RP2040 board. And I tried to connect via OpenOCD, but nothing worked for me. What am I doing wrong?

`C:\Users\Professional>openocd -f interface/cmsis-dap.cfg -c "adapter speed 400" -f target/stm32f1x.cfg
xPack Open On-Chip Debugger 0.12.0+dev-01850-geb6f2745b-dirty (2025-02-05-09:32)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 400 kHz
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E4612C440B494E25
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Error: Error connecting DP: cannot read IDR `

P33M, 2025-05-06:
Your debugprobe is working correctly. However we're not going to offer support for connection to third party target boards that we know nothing about; this is where reading your vendor's datasheets is required.

lurch, 2025-05-06:
See also #172

closed 2025-05-06 by lurch