raspberrypi/debugprobe #175: RTT is marked "TODO" but works for me

eliasnaur, 2025-05-24:
The [README](https://github.com/raspberrypi/debugprobe/blob/master/README.md#todo) states:

> TODO
> Possibly include RTT support

However, I've successfully used the probe with [TinyGo](https://tinygo.org) in RTT output mode with success.

It seems to me the TODO is fixed, or perhaps I missed something?

eliasnaur, 2025-05-27:
What does a log look like? I ran `tinygo -target pico2 -serial=rtt -monitor ... ` and saw output from my program running on a rp2350, through a pico probe.

JonasFocke01, 2025-06-06:
Im using a pico with the firmware from this repo as a debug probe, and a second pico to run my actual software. RTT works very well for me. Ive tested the pico 1 and 2 in all combinations. Works flawless. Im using [cargo embed](https://probe.rs/docs/tools/cargo-embed/) with absolute default config, together with [rtt_target](https://docs.rs/rtt-target/latest/rtt_target/), with also default configuration. Using `rprintln!("...")` shows up just fine in my host-os-terminal.

closed 2025-07-16 by P33M